f4exb/sdrangel #2524: Version 7.22.9 crashes immediately on MacOS Sequola arm64

thomasgi1, 2025-09-28:
When starting SDRangel on MacOS Sequola it crashes due to a code signing problem:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SDRangel [33640]
Path:                  /Applications/SDRangel.app/Contents/MacOS/SDRangel
Identifier:            SDRangel
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-09-27 10:45:21.9450 +0200
OS Version:            macOS 15.6.1 (24G90)
Report Version:        12
Anonymous UUID:        5C82BB8A-F941-91DF-82B5-591D17C1D027

Sleep/Wake UUID:       AB4ACB97-5348-43DD-A5E7-AEBB6E07E4F8

Time Awake Since Boot: 1100000 seconds
Time Since Wake:       3420 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Codes:       UNKNOWN_0x32 at 0x000000011e3bc000
Exception Codes:       0x0000000000000032, 0x000000011e3bc000

Termination Reason:    Namespace CODESIGNING, Code 2 Invalid Page

VM Region Info: 0x11e3bc000 is in 0x11e3bc000-0x11e3e8000;  bytes after start: 0  bytes before end: 180223
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      __LINKEDIT                  11e3b8000-11e3bc000    [   16K] r--/rwx SM=COW  /Applications/SDRangel.app/Contents/Frameworks/libmbe.1.dylib
--->  mapped file                 11e3bc000-11e3e8000    [  176K] r--/rwx SM=COW  Object_id=b342fe00
      GAP OF 0xc000 BYTES
      __TEXT                      11e3f4000-11e4d0000    [  880K] r-x/rwx SM=COW  /Applications/SDRangel.app/Contents/Frameworks/libft8.dylib

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   dyld                          	       0x187e742b8 mach_o::Header::forEachLoadCommand(void (load_command const*, bool&) block_pointer) const + 48
1   dyld                          	       0x187e77214 mach_o::Header::forEachSegment(void (mach_o::Header::SegmentInfo const&, bool&) block_pointer) const + 124
2   dyld                          	       0x187e76644 mach_o::Header::preferredLoadAddress() const + 124
3   dyld                          	       0x187e23290 dyld4::JustInTimeLoader::withRegions(dyld3::MachOFile const*, void (dyld3::Array<dyld4::Loader::Region> const&) block_pointer) + 52
4   dyld                          	       0x187e23ac8 dyld4::JustInTimeLoader::makeJustInTimeLoaderDisk(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int, mach_o::Layout const*) + 1108
5   dyld                          	       0x187e168b0 dyld4::Loader::makeDiskLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int, mach_o::Layout const*) + 176
6   dyld                          	       0x187e182cc invocation function for block in dyld4::Loader::getLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&) + 2884
7   dyld                          	       0x187e16fd8 dyld4::Loader::forEachResolvedAtPathVar(dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, dyld4::ProcessConfig::PathOverrides::Type, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 764
8   dyld                          	       0x187e01d60 dyld4::ProcessConfig::PathOverrides::forEachPathVariant(char const*, mach_o::Platform, bool, bool, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) const + 556
9   dyld                          	       0x187e16b18 dyld4::Loader::forEachPath(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 260
10  dyld                          	       0x187e173b0 dyld4::Loader::getLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&) + 920
11  dyld                          	       0x187e216d4 invocation function for block in dyld4::JustInTimeLoader::loadDependents(Diagnostics&, dyld4::RuntimeState&, dyld4::Loader::LoadOptions const&) + 456
12  dyld                          	       0x187e762f4 invocation function for block in mach_o::Header::forEachLinkedDylib(void (char const*, mach_o::LinkedDylibAttributes, mach_o::Version32, mach_o::Version32, bool, bool&) block_pointer) const + 136
13  dyld                          	       0x187e74358 mach_o::Header::forEachLoadCommand(void (load_command const*, bool&) block_pointer) const + 208
14  dyld                          	       0x187e75fe0 mach_o::Header::forEachLinkedDylib(void (char const*, mach_o::LinkedDylibAttributes, mach_o::Version32, mach_o::Version32, bool, bool&) block_pointer) const + 160
15  dyld                          	       0x187e213e4 dyld4::JustInTimeLoader::loadDependents(Diagnostics&, dyld4::RuntimeState&, dyld4::Loader::LoadOptions const&) + 164
16  dyld                          	       0x187e21470 dyld4::JustInTimeLoader::loadDependents(Diagnostics&, dyld4::RuntimeState&, dyld4::Loader::LoadOptions const&) + 304
17  dyld                          	       0x187e3f5f8 dyld4::APIs::dlopen_from(char const*, int, void*)::$_0::operator()() const::'lambda'()::operator()() const + 500
18  dyld                          	       0x187e3f390 void dyld4::RuntimeLocks::withLoadersWriteLockAndProtectedStack<dyld4::APIs::dlopen_from(char const*, int, void*)::$_0::operator()() const::'lambda'()>(dyld4::APIs::dlopen_from(char const*, int, void*)::$_0::operator()() const::'lambda'())::'lambda'()::operator()() const + 192
19  dyld                          	       0x187e3edb4 dyld4::APIs::dlopen_from(char const*, int, void*)::$_0::operator()() const + 944
20  dyld                          	       0x187e33f9c dyld4::APIs::dlopen_from(char const*, int, void*) + 1136
21  dyld                          	       0x187e33a80 dyld4::APIs::dlopen(char const*, int) + 128
22  QtCore                        	       0x109027348 0x108d14000 + 3224392
23  QtCore                        	       0x109022760 0x108d14000 + 3204960
24  QtCore                        	       0x1090223f0 0x108d14000 + 3204080
25  QtCore                        	       0x108e24e94 QPluginLoader::load() + 124
26  libsdrbase.dylib              	       0x1064dd8f8 PluginManager::loadPluginsDir(QDir const&, QList<QString> const&) + 416
[... 227 lines cut ...]
       0x11df24000 -        0x11df37fff org.qt-project.QtSerialPort (6.9) <f88cd019-e95b-3ec8-bb08-cdc64f89f245> /Applications/SDRangel.app/Contents/Frameworks/QtSerialPort.framework/Versions/A/QtSerialPort
       0x187df8000 -        0x187e93577 dyld (*) <3247e185-ced2-36ff-9e29-47a77c23e004> /usr/lib/dyld
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???
       0x188199000 -        0x1881a5a47 libsystem_pthread.dylib (*) <d6494ba9-171e-39fc-b1aa-28ecf87975d1> /usr/lib/system/libsystem_pthread.dylib
       0x18815d000 -        0x188198653 libsystem_kernel.dylib (*) <6e4a96ad-04b8-3e8a-b91d-087e62306246> /usr/lib/system/libsystem_kernel.dylib
       0x193811000 -        0x193838ddf com.apple.audio.caulk (1.0) <42085f32-42e2-3f11-b0b4-0343137b5f72> /System/Library/PrivateFrameworks/caulk.framework/Versions/A/caulk
       0x18820e000 -        0x18874cfff com.apple.CoreFoundation (6.9) <8d45baee-6cc0-3b89-93fd-ea1c8e15c6d7> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
       0x18c173000 -        0x18d603e3f com.apple.AppKit (6.9) <860c164c-d04c-30ff-8c6f-e672b74caf11> /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
```

However, when I code-sign the application by myself, it starts without crash, but does not recognise my HackRF.

loewal, 2025-09-29:
Same here. Installed the Intel-version, works, but no sdrplay-support.
Reinstalled Arm v8, works, with sdrplay-support.

Axpelle, 2025-09-29:
I can confirm that the crash on opening is happening in my case as well.

loewal, 2025-09-30:
> 1. Remove quarantine flag: `xattr -dr com.apple.quarantine Applications/SDRangel.app/`
> 2. Configure ad-hoc signatures for binaries: `codesign --force --deep --sign - Applications/SDRangel.app/`

Can confirm also.
Thanks!

WinkelCode, 2025-09-30:
@Axpelle this can't be fixed in code\* as both of the commands are related to macOS security measures (gatekeeper) and are specific to a macOS install.

\*However, the .dmg could be shipped with an install script that automates this. I believe they're commands don't require elevated permissions.

srcejon, 2025-09-30:
It seems to be caused by any plugin that uses libusb.

srcejon, 2025-09-30:
> > does not recognise my HackRF.
> 
> The HackRF library has been updated from v2024.02.1 to the latest git revision, so probably something to do with that.

From the log:

Cannot load library /Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputhackrf.dylib: dlopen(/Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputhackrf.dylib, 0x0085): Library not loaded: @rpath/libhackrf.0.dylib

Bundle contains:

./Contents/Resources/lib/plugins/libinputhackrf.dylib
./Contents/Resources/lib/plugins/liboutputhackrf.dylib
./Contents/Resources/lib/libhackrfdevice.dylib
./Contents/Resources/lib/libhackrf.0.dylib
./Contents/Resources/lib/libhackrf.dylib
./Contents/Resources/lib/libhackrf.0.9.0.dylib
./Contents/Frameworks/libhackrfdevice.dylib

srcejon, 2025-10-01:
> It seems to be caused by any plugin that uses libusb.

Replacing libusb-1.0.0.dylib with the binary from the previous release fixes the problem. We didn't actually change the version of libusb source used though, so perhaps something in the build environment. A newer version of libusb is available.

Axpelle, 2025-10-01:
Tx for the info. Is a refreshed version with the previous binary in the works? Or do we need to wait for 7.22.10?

srcejon, 2025-10-01:
> Tx for the info. Is a refreshed version with the previous binary in the works? Or do we need to wait for 7.22.10?

I don't know what the fix is yet - and the github Mac runners don't appear to be working at the moment either, so can't test.

hiarmen, 2025-10-02:
In my case app starts, but no more loading the HackRF One.

Axpelle, 2025-10-05:
> It seems macdeployqt is corrupting the signature some how.
> 
> But we can do adhoc resigning before cpack creates the dmg.
> 
> Try this: https://github.com/srcejon/sdrangel/actions/runs/18244685074/artifacts/4182431315

Glad to report that you nailed it. I'm giving it a test run to check its stability. Tx so much for the fix.

closed 2025-10-31 by f4exb