f4exb/sdrangel #2526: FLATPAK: After updating SDRAngel from version 7.22.8 to 7.22.9, SDRAngel no longer starts.

DL3BCO, 2025-09-30:
After updating SDRAngel from version 7.22.8 to 7.22.9, SDRAngel no longer starts.

$ /usr/bin/flatpak run --branch=stable --arch=aarch64 --command=sdrangel org.sdrangel.SDRangel
2025-09-30 14:41:05.565 (W) Qt: Session management error: Could not open network socket
2025-09-30 14:41:05.601 (I) SDRangel 7.22.9 Qt 6.8.3 64b arm64 KDE Flatpak runtime DSP Rx:24b Tx:16b PID 2

Operating System: Kubuntu 25.04
KDE Plasma Version: 6.3.4
KDE Frameworks Version: 6.12.0
Qt Version: 6.8.3
Kernel Version: 6.14.0-1013-raspi (64-bit)
Graphics Platform: X11
Processors: 4 × ARM Cortex-A76
Memory: 7.7 GiB of RAM
Graphics Processor 1: V3D 7.1.7.0
Graphics Processor 2: llvmpipe
Product Name: Raspberry Pi 5 Model B Rev 1.0�

Axpelle, 2025-09-30:
We're having the same issue on macOS Sequoia with ver 7.22.9. I have reverted to ver 7.22.8 pending a fix.

AsciiWolf, 2025-11-21:
I can't. Qt is a part of the Flatpak runtime. I would have to compile the whole Qt module myself. And I still have no way to test the result myself since I have no ARM computer here.